mvandersteen/SungrowInverter #1: SG15RT

dermobb, 2022-01-08:
Hi and thanks for your impressive work!

I'm about to receive a SG15RT string inverter myself in a few weeks. Can you confirm your script works with it? I know you listed it as supported, but I read elsewhere that Modbus TCP does currently not work on string inverters, only on the hybrid ones. Thank you.

hallonstedt, 2022-03-10:
It works on string inverters. My SG15KTL is happily sending data.

Make sure you have python 3.9+ and `pip3 install SungrowInverter`
Edit the IP address in [test.py](https://github.com/mvandersteen/SungrowInverter/blob/main/test.py) and you're good to go.

mvandersteen, 2022-05-29:
Very delayed response sorry @dermobb I can't confirm any other than the hybrid SH5K-v13 that I have.

I have used the documents found online to support the string inverters as well and can only hope it helps others.

If you look at the TI_20210716_Communication Protocol of PV Grid-Connected String Inverters_V1.1.37_EN.pdf document which refers to your particular inverter, on page 4 is reference to optional TCP support. And this is what i have used to allow this to support these type of inverters.

You may need to purchase a supporting wifi (Winet-s ny the looks) dongle to plug into the inverter to support network communications, which looking at a [manual](https://www.manualslib.com/manual/2106910/Sungrow-Sg5-0rt.html?page=47#manual) for your inverter indicates is required. Best to talk to your installer or local Sungrow rep and they will point you in right direction though. Sorry I can't offer concrete support.

closed 2022-05-29 by mvandersteen