mvandersteen/SungrowInverter #3: Sungrow parallel TCP & RTU

Hantuch, 2022-04-27:
Hello there, has anyone tried running both the Modbus RTU & Modbus TCP from Sungrow's RS-485 & Ethernet port respectively?

Will it interfere with each other?

mvandersteen, 2022-05-08:
@Hantuch, I've no idea on this sorry. 

I've only ever used MODBUS via TCP on my particular inverter Sungrow SH5K, so cannot comment on this, would advise to google on other forums where modbus is discussed.

While not completely on topic, I have had issues with the old Sungrow wifi device (v24) that i have on my inverter and connecting to a Asus Wifi-6 router. That's been very painful and why I've not bothered with this project much lately. I've since added a raspberry pi to my network to act as a access point for my IOT devices (like) this and works much better and stable for connections from my router. So will continue to develop further shortly

closed 2022-05-08 by mvandersteen

Hantuch, 2022-05-10:
Aight thanks

I'll be using the RS-485 port on the SG40CX to get data from it, basically replacing a Solarlog 50 logger that were used previously with it. If it works with Solarlog, it should theoretically works other my custom logger since the Modbus RTU protocol is rather simple (i hope)

Well wish me luck!