mvandersteen/SungrowInverter #4: Sungrow Device Model: SG5K-D

Stephen2615, 2022-10-23:
Greetings

I have a Sungrow Device Model: SG5K-D with a Wifi V31 dongle on it.  When trying the test.py, I get this error. 

ERROR:UPSUPPORT INVERTER: Supported inverter device_type_code [294] is not supported

I see that someone has a fix for this with a HA docker but I  haven't implemented it yet. 

Regards

Stephen

mvandersteen, 2022-11-26:
You could try adding the following to the end of the inverter list.

SungrowInverterModel(0x126,'SG5K-D','string',2),

I found a manual for your SG5K-D inverter but those inverter were not part fo the modbus doc's i had so , can't say if that will work or not.
note the 294 is decimal for the code which are stored in the list of inverters hex value is 0x126 and that 294 was not included hence the error.
I'll be interested how you trave with this one.

closed 2022-11-26 by mvandersteen

Stephen2615, 2022-11-29:
Thanks. It did work.

***@***.***:~/SungrowInverter $ python3 test.py
INFO:Sungrow residential inverter found: [Device Code: 0x126, Model: SG5K-D, Nominal Ouput: 5.0 kW,

Stephen2615, 2022-11-29:
Greetings,

I added that string to the end and it works very well. Thank you for this help.

Cheers

Stephen