mvandersteen/SungrowInverter #5: Missing and broken dependencies

karottenreibe, 2022-11-01:
Tried installing on python 3.10.5. This failed due to:

- pyserial-asyncio is not listed as a requirement
- pymodbus is installed in version 3.0.0 but it seems an older version is required

This fixed it for me:

```
pip install pyserial-asyncio
pip install pymodbus==2.5.3
```
It would probably best to update requirements.txt to list the missing package and pin the pymodbus package to the desired older version.

mvandersteen, 2022-11-26:
This is due to the updates required for pymodbus version 3.
The maintainer of SungrowModbusTCPClient has made the changes necessary to his package but its not been updated on PyPI yet so will have a look and see what I can do to get this working, without reliance on that package for now.

mvandersteen, 2022-12-12:
Fixed while waiting for release of 0.1.7 of SungrowModbusTCP to PyPi. 
Hopefully no breaking changes in future to pymodbus

closed 2022-12-12 by mvandersteen