mvandersteen/SungrowInverter #6: Division by zero error early in day when pv generation is zero

smnrgrs, 2022-11-19:
I think I'm seeing a division by zero error from Line 149 of Hybrid.py early in the day before there's been any generation. As eval() only takes simple expressions a fix may be to add a small quantity to daily_pv_generation, e.g. 0.001

mvandersteen, 2022-11-26:
Wrapped in a try catch now, and outputs 0 if an error occurs should be resolved for next release.
Just working through issue with Modbus updates at the moment.

smnrgrs, 2023-02-05:
Is there any way to force HA to update- still suffering this big.

Octofinger, 2023-03-13:
I think I found an issue with that fix. The eval of calculated registers is only followed by a finally that always set the result to zero.
Instead you should catch ZeroDivisionError and then set the result to zero. Skip the finally clause entirely.

tinywiz, 2024-01-12:
Any update on this issue? Having the same problem.

smnrgrs, 2024-01-12:
I also still have the issue. If I could modify the running SungrowInverter python script on my HA pi I could probably find a fix, but I can't find where the libraries are.

tinywiz, 2024-01-13:
> I also still have the issue. If I could modify the running SungrowInverter python script on my HA pi I could probably find a fix, but I can't find where the libraries are.

I decided to run it locally on a n ubuntu server, modified the code and it works now :)

smnrgrs, 2024-01-13:
Great, would you be able to summit a pull request?

tinywiz, 2024-01-13:
> Great, would you be able to summit a pull request?

Honestly, I have no clue on how to do that :(