mvandersteen/SungrowInverter #7: it's not a bug ! Thanks a lot.

jarod1990-Developper, 2022-12-14:
Thanks a lot for this work. If you want you can add this inverter :
    SungrowInverterModel(0x2600,'SG2.0RS-S','string',2),

congratulations!

mvandersteen, 2023-01-05:
This one is a strange one, most dev work is on a dev container so using Debian so don't believe linux is the issue.
What OS are you running on the Pi, I'll then test on my pi as well?

jarod1990-Developper, 2023-01-14:
Good afternoon,
sorry for the delay,

My OS is
[image: image.png]
thanks for your help.

BurhanFathur, 2023-06-12:
hi everybody.
I encountered a problem like this. How do I solve it? Please help
[image]

closed 2024-05-17 by mvandersteen