mvandersteen/SungrowInverter #8: My SH10RT is 3599 (0xE0F) instead of 3587 (0xE03)

adentse, 2023-01-30:
Changing E03 to E0F in inverter.py makes it work a bit better but something happens:

Running test.py with debug gives this:

http://adent.chroma.se/output.txt

Best regards
Mikael

Octofinger, 2023-03-13:
Got the same issue as you. I have a SH10RT-V112 that identifies as 0xE0F (3599). I will see if mr @mvandersteen would allow me to do some updates to this repository to correct that. ¨
I had to add the identifier to the inverter.py list of inverters, but also to the ModBusRegisters list in hybrid.py so three phase results are returned to this inverter as well.
The division by zero issue you see there is reported in another issue and is not related to your missing inverter.

mvandersteen, 2023-03-22:
Sorry for delay, clearly not been on for awhile... 

I've accepted the merge request, and will sort out the python package after this.

After thinking about this request though i think i need to add the older code back in, potential for "different" version inverter having the E03 code as well. although who knows what it will be will update the inverter in this case to a different name with the version tacked onto the end, cover both what your seeing and what documentation has as well.

closed 2024-05-17 by mvandersteen